parallaxinc/Parallax-ESP #2: Request for Invalid Handle/ID Error Code and Response

PropGit, 2016-06-21:
On the Serial Interface, when calling a command like PATH with an invalid _handle_ or _id_, the response is CMD_ERROR_INVALID_STATE.

Can we change this condition to respond with an error that means INVALID_HANDLE_ID (an error code that doesn't yet exist) instead of INVALID_STATE?

PropGit, 2016-06-22:
<<< ON HOLD >>>
It was later revealed that INVALID_STATE was an appropriate response for the specific case involved.

I'm still thinking we need an INVALID_HANDLE_ID error code as a possibility, but I will follow-up on this and either mark it closed, or indicate the continued need for it at that time.

dbetz, 2016-06-22:
There are currently a lot of errors related to arguments that are just reported as "invalid argument". That error could be split into multiple error codes that are more specific. I just used "invalid argument" because it roughly parallels the Linux/Unix INVAL error code that means the same thing.

closed 2018-03-22 by PropGit