parallaxinc/Parallax-ESP #3: Add "Home" Link to Each Web Page (other than the Home page)

PropGit, 2016-06-21:
On the Wi-Fi Networks page
[image]

... Upload page
[image]

... Settings page
[image]

... and the Firmware Update web page,
[image]

Please add a "Home" link, or button, that takes the user back to the home page.
[image]

dbetz, 2016-06-22:
NP. Do you have a graphic you'd like to use for the button?

dbetz, 2016-06-28:
I uploaded a firmware update that puts a navigation bar on the left side of the screen that will allow you to jump from any page to any other page including the home page.

PropGit, 2016-06-29:
I like the navigation bar effort but have yet to try it.  I'm concerned about responsive layout on constrained screens (smartphones in portrait orientation, especially).

If the nav bar makes it difficult to use on constrained screens, short of adding responsive layout features, we'd probably opt for a single link.  Before you added the nav bar, I asked our web/graphic artist for input and she suggested the following two possibilities that I'm adding here just in case we need to revert to it.  See lower-right of each image.

In the meantime, no need to make any more changes until I test and respond with an additional post.

[image]

[image]

PropGit, 2016-10-13:
Closing because I believe this issue has been fully addressed now.

closed 2016-10-13 by PropGit