parallaxinc/Parallax-ESP #4: Should we have a cmd-end-char setting?

PropGit, 2016-06-24:
[This is extracted from a conversation in the Parallax Wi-Fi Module API document.  It is placed here to be decided on later.]

__PropGit__
Future thought: Since we have _cmd-start-char_, shouldn't we also have _cmd-end-char_?

__dbetz__  I asked @AndyLindsay about this when we added cmd-start-char and he said he preferred to always use \r.

__AndyLindsay__
I think an optional cmd-end-char is fine, but we want the default to be \0x0D.  Also, if certain terminals use \n\r, it might be worth adding that as an option too.

__dbetz__
If we allow a CR/LF or LF/CR option then the command must end with both characters. It's more difficult to allow either CR or CR/LF for example. Also, should I be able to set the end character to "?" or something like that? In fact, it's impossible to set it to ":" or "," since those characters are already used in commands. For that matter "/" is used in paths. Maybe the options should only be CR, LF, CRLF, or LFCR. In fact, I don't know of any terminals that use LFCR so that can probably be left out.

PropGit, 2016-06-24:
__Thoughts__
The Serial interface will most often be used with an attached microcontroller, so single character endings will continue to be common there.

It's a very interesting thought to attach a terminal to those connections; it'd help foster experimentation that's very important for education.  Most terminals output a two-character line ending.

Having options for two-character endings makes the setting name incorrect (cmd-end-char), so we'd probably want to adjust the name, and I think we should limit it to a set of CR (the default), LF, and CRLF.

dbetz, 2016-06-24:
While I'm happy to do what you want I don't think that two-character line ends are common with terminal programs. I think the most common line end is CR if it is really emulating an old-style hardware terminal or maybe LF. CRLF is probably the least common and may not even be supported by some terminal emulators. Do you know of one that requires or even defaults to CRLF?

PropGit, 2016-10-13:
To follow-up on this, we _should_ have a settable end character, but just a single character. Default should be ASCII 13 "\r".

dbetz, 2016-10-13:
Out of curiosity, why do you think that’s needed? My theory is that the start character gets you into command mode and the user might want to select which character triggers that because the one we choose might interfere with something the user would like to use in transparent mode. However, the end character gets us out of command mode and the syntax of commands is entirely defined by us. What would be the point of changing the end character?

PropGit, 2016-10-13:
Just for the same reason that the start character is configurable; to be flexible for different applications that may have a compelling reason to use a different character.  In the Docs, I wrote that they are both configurable and indicate them in syntax descriptions by a single character symbol so that we can stay away from obscuring the meat of the command or response further with multi-character representations of the start and end character... so a configurable end character fits nicely within the context of the docs also.

dbetz, 2016-10-13:
As I mentioned, there is no need to configure the end character because it is part of our CMD syntax and not an escape from normal user text. in fact, I’m not sure changing either is really a good idea since it breaks all of the sample code.