parallaxinc/Parallax-ESP #6: AP > STA Mode Switch

PropGit, 2016-10-13:
@AndyLindsay - Let's rethink the Networks web page interface with regards to Wi-Fi Mode switching and setting the passphrase.

# Background
Currently, selecting a different mode from the Wi-Fi Mode dropdown field changes the mode immediately.  This has benefits, but the options we provide lead to problems in certain cases.

## When viewing Networks page through the AP (SoftAP) interface
From the user's perspective,  switching from AP to STA+AP is a three-step process:
  1. Change Wi-Fi Mode field from AP to STA+AP
  1. Select the desired access point (SSID) and enter it's passphrase
  1. Click the Connect button

The active mode switch happens immediately on step 1 (perhaps unbeknownst to the user) but because the point of that mode switch is to connect the module's STA interface to another access point, steps 2 and 3 must be performed to complete the process.  __This makes sense__.

[image]

## When viewing Networks page through the AP (SoftAP) interface
Switching from AP to STA occurs immediately; however, the module does not remember any previously entered passphrase for any particular SSID, and SSID scanning and page update is not performed in this mode- it is, of course, now disconnected from the very browser that was viewing its Networks page.  This means that switching from AP to STA leaves the module stranded; no AP interface and the STA interface is active but not connected to any access point.  __This _doesn't_ make sense to allow__.

[image]

## When viewing Networks page through the STA interface
Switching from STA to STA+AP mode, or from STA+AP to STA mode makes sense, happens immediately, and doesn't leave the module stranded.  The only oddity is that __SSID scans are not performed unless it's in STA+AP mode__.

Switching from STA to AP mode, or from STA+AP to AP mode, happens immediately and doesn't leave the module stranded- it can be reached from a device via it's AP interface.  __No problem here__.

# Proposed Solution
Ideally, it would be best to ensure the user cannot "strand" the module through an innocent action over the configuration web pages.  The Networks page allows changing from AP to STA mode, stranding the module.

Also, the SSID scanning feature is annoying (similar to other interfaces) because of how slow it is.  Instead of it being automatic, we should make the user manually click a "Scan Now" button, and provide them visual feedback that the scan is in process, before finally showing them the results.

So... 
  - if it's in AP mode, the configuration page should _only_ allow switching to STA+AP mode (ie: not STA mode).
  - if it's in STA+AP mode, the configuration page should only allow switching to STA mode if, and only if, the STA interface already has an IP address (it's associated).
  - if it's in STA mode, the configuration page should allow switching to either STA+AP or AP (it, of course, already has an IP address (it's associated) in this case).
  - It should not automatically scan for SSIDs, but rather should only scan upon the user clicking the Scan Now button.  It should ideally show an animated busy indicator from then until the point that the found SSID list is displayed or the user navigates away from the Networks page.

@AndyLindsay @dbetz - Please consider this for the second release of firmware.

PropGit, 2016-10-17:
Another thing to consider is user feedback for Wi-Fi Mode changes.

PropGit, 2016-11-03:
Along with this, Issue #11 must be considered and implemented as well.

dbetz, 2017-02-13:
What is the status of this request? The first message says "Do not perform the firmware changes noted in this thread". Is that still true or is this ready for implementation? Would the Parallax web design team like to implement the browser side of this?

PropGit, 2017-02-13:
Other tasks are higher priority at the moment and requires some of my time to drive it.  This isn't ready to be resolved yet, but we don't want to lose the request.