parallaxinc/Parallax-ESP #7: SoftAP Security

PropGit, 2016-10-13:
We'd like to add a security feature to the AP interface (SoftAP).

It should default to Open (as it is now) but should allow for setting to WPA2 (which includes a passphrase that must also be settable).

PropGit, 2016-12-23:
We'd also like to have a way to reset it back to a default (no passphrase; open) with physical contact of the module in case the user forgot what the password is; i.e.: last-ditch recovery.

dbetz, 2017-02-12:
Can the way to switch it back to open be through a serial command? Or does it have to be something that can be done by pressing the reset button in a particular pattern like we currently use to go to SoftAP mode?

PropGit, 2017-02-13:
A serial command would be fine; however, you've got me thinking about a reset-button option.

@AndyLindsay - What would you think about another reset-button option as a physical-access safety net for a user to wipe the passphrase and switch back to OPEN SoftAP mode?  For example, like pressing/releasing the reset button 10 times at a brisk pace?  Or pressing/releasing 6 times but holding down for greater than 2 seconds on the 6th?  Something not super likely to be done by accident.

AndyLindsay, 2017-02-13:
How about holding the RST button for 10 s followed by 6 taps?  My reasoning
is that students will tend to ta the RST button more than 4x, so this would
be a more unique sequence that they'd b less likely to do just because they
are fidgeting.

Andy

PropGit, 2017-02-13:
Excellent!

@dbetz  - Please add a reset button method to clear the SoftAP passphrase and return the module to Open mode on SoftAP.  The algorithm is:
  1) Is SoftAP enabled?  If not, ignore # 2.
  2) If user holds reset button for > 10 seconds, then,
    2a) releases reset button for no longer than 2 seconds, then
    2b) presses/releases reset button 6 times swiftly (all 6 press/release cycles occur in less than 3 seconds), then, and only then, perform # 2c.
    2c) Reset to Open Soft AP as indicated below
      - Eject any current connections to the Soft AP interface (if this doesn't happen automatically upon mode change),
      - clear SoftAP's passphrase,
      - set to Open SoftAP mode (if necessary),
      - and perform any other needed operations for a clean switch to Open Soft AP mode.

PropGit, 2017-02-13:
@AndyLindsay - What do you think... is a Serial API for this really needed?  If someone has network access already, then they can reprogram the Propeller remotely to cause the module to disable local SoftAP security... all without physical access to the module.  Of course, if they have network access already, there are other malicious things they can do, but disabling SoftAP security seems like something that should require physical access.

Hmm... right now, they can change SoftAP security (once we have it) over the network via configuration page, so maybe that concern is too hard to justify.  I guess this only becomes a question with the thought of the possible future feature of password-protecting the configuration pages.

Please advise.