parallaxinc/Parallax-ESP #9: Can DBG be disabled by default?

PropGit, 2016-10-13:
@dbetz - Can you think of a way that we could set DBG to be optional and disabled by default?  Some way that doesn't prevent us from using DBG for our uses (development) in all the cases we need it to be?

dbetz, 2016-10-13:
Unfortunately, most of those messages are printed by os_printf which, as far as I know, isn’t switchable. They should probably all be switched over to some new function like dbg_printf that maybe would take an extra argument indicating the type of message along with a way to select specific messages to print. That way you could, say, enable all CMD handling messages or all wi-fi status messages, or all HTTP messages, etc.

PropGit, 2016-10-13:
Okay, let's consider this for Release 2, among others.

PropGit, 2016-10-13:
__Update__ - Andy determined that he could set the DBG to input via a command at the start of the PBASIC example code... that allows us to use P12 in the BOE for servo control and it won't be affected by DBG.  

So, the ability to filter DBG output is how this issue can proceed, but it's purely helpful for development purposes and no longer an issue for our tutorials or for users.  It can proceed at a pace conducive to it's need for us developers.

dbetz, 2017-02-13:
This is done. There is a setting that enables/disables output to the DBG pin. The default is "disabled". There is also a "Debug Log" page in the web interface that shows the most recent 1400 bytes of debug output.

closed 2017-05-08 by PropGit