parallaxinc/Parallax-ESP #12: Add commands for SSID Scan results

PropGit, 2016-11-07:
Add Network and Serial commands to scan for SSIDs.  The intent is that this returns a string containing multiple substrings, each of which is an SSID that it found during a scan.

dbetz, 2017-01-06:
There is already a network API for scanning for access points. That is what the configuration web page uses.

dbetz, 2017-01-08:
To add a bit more information, the network API for initiating a scan and returning discovered access points is "/wifi/wifiscan.cgi". It returns a JSON response that contains the access points discovered on the previous call to "/wifi/wifiscan.cgi". The format of the JSON reply is:

When the scan is still in progress:

{
"result": {
  "inProgress": 1
  }
}

When the scan completes:

{
"result": {
  "inProgress": 0,
  "APs": [
    { "essid": "x",
      "bssid": "x",
      "rssi": "x",
      "enc": "x",
      "channel": "x"
    },
    (more entries as above)
    ]
  }
}

Where "x" above indicates the value of the associated field.

dbetz, 2017-01-13:
Is the existing network API for scanning as described above okay or do we need a different network API?

PropGit, 2017-01-18:
> Is the existing network API for scanning as described above okay or do we need a different network API?

The existing network API is fine as-is.

PropGit, 2017-01-18:
UPDATE: So far we've settled on the commands APSCAN (to start the Wi-Fi Scan process) and APGET (to retrieve the found Wi-Fi SSIDs).

PropGit, 2017-01-18:
@dbetz - Maybe we should think about this a bit more.  Would it be possible to filter the results by encryption type or simple search strings, like "xyz," which would return SSIDs _Wxyz_, and _AbcXYz_, but not _Linksys_, or _FreeWiFi_, for example.

dbetz, 2017-01-18:
Actually the ESP API allows some filtering that doesn't get used currently. They can filter by SSID, BSSID, channel number, or whether the AP is hidden. There is no pattern matching on SSID though.

PropGit, 2017-01-19:
UPDATE: v2017-01-19-0828 includes the APSCAN and APGET Serial API commands.

__APSCAN__ 
  * takes no arguments
  * returns "S,0"
  * if events are enabled, an event “A,count,0” arrives when the scan completes

__APGET__ 
  * takes a single argument, a numeric value from 0 to N-1 where N is the number of access points found during the scan.
  * returns “N,0” if the scan is not yet complete
  * returns "E,code" if an out-of-range value is given
  * returns “S,enc,ssid” if the scan has completed.  The contents relate to the item given by the APGET argument.
    * enc is an enumerated value indicating the security encryption type
    * ssid is the SSID string

dbetz, 2017-02-12:
Can this be closed?

PropGit, 2017-02-13:
@AndyLindsay - This feature is now in the new firmware as described above.  Does this meet your needs?  Have you had a chance to test it yet?

AndyLindsay, 2017-02-14:
Thank you, I have not been able to test it.  I'm having some computer
hardware problems, and everything is going at a snail's pace right now.

AndyLindsay, 2017-02-14:
I just got another report of the same symptom.

Any info you can get from them about OS, browser, setup, security settings
and security software that might be causing the problem would be
appreciated.

PropGit, 2017-02-16:
> I just got another report of the same symptom.

@AndyLindsay - Was that post really intended for this issue?

PropGit, 2017-05-02:
Leaving this issue for @AndyLindsay Andy to test and respond when he has time.  Please report and close the issue (if tested successfully).

PropGit, 2017-05-08:
We don't anticipate problems with this; closing the issue.

closed 2017-05-08 by PropGit