parallaxinc/Parallax-ESP #14: Name to SSID: Verify and add rules

PropGit, 2016-11-28:
Verify that Module Name changes are being reflected as the SSID.  This seems to work sometimes and not others (PropGit witnessed differences between using his workstation and a tech support workstation but has not retested).

Apply PropLoader's module name rules to firmware to limit name (and thus SSID) to safe characters.

dbetz, 2016-12-23:
Have you tried this with the latest firmware? A forum member reported this a while back and I thought I had fixed it.

PropGit, 2016-12-24:
I have not... but I have a faint memory of what you're describing too.  One of us will have to check it out before resolving this item.

PropGit, 2017-01-13:
I will have to retest this using my system at home office and tech support's system at Rocklin office to verify.

dbetz, 2017-02-12:
Did this ever get tested? Can it be closed?

PropGit, 2017-02-13:
@dbetz - No it didn't, but I can test tomorrow morning using the same scenario as before.  I will make the effort to do so.

PropGit, 2017-02-21:
Verified that this works now.  Tried "Save to Flash" each time.

closed 2017-02-21 by PropGit