parallaxinc/Parallax-ESP #16: Add ability for secure communication with IFTTT.

PropGit, 2017-01-10:
Add ability to have some form of secure data exchange with ifttt.

PropGit, 2017-01-10:
[From AndyLindsay] : currently, a user can grab a string that triggers an ifttt event from their browser's inspect tool, remove the 's' from "https" in the POST/GET request strings, then send it over TCP with the Wi-Fi module, and it still works.  ...but, we don't know when that non-secure support might evaporate.

dbetz, 2017-09-29:
Are you asking for the WX to be able to act as an SSL client? That might be possible but being an SSL server would be hardware. Also, most of the posts I've seen related to SSL on the ESP8266 mention that it requires lots of RAM. Not sure if we have enough free RAM for this.