parallaxinc/Parallax-ESP #17: Reset Signal over RTS doesn't work without help

PropGit, 2017-01-31:
It appears that the Parallax-ESP firmware initializes the RTS pin to low.  This may be correct (please verify).

However, when the reset signal is redirected to RTS, the RTS pin starts out low and stays low.

We need the RTS pin to rest high when the reset signal is directed there, then it should toggle low and high to cause the reset.

Here's what @AndyLindsay reported after he tested it-
> __[From Andy's notes]:__ I used the serial API to set /RTS high. Then, I moved the jumper wire from /DTR to /RTS, and the program load was successful. I then cycled power, and went back to output-low.,
...and unable to load the program.
I used what SimpleIDE generated to run the program that serially set /RTS to output-high. After that, I successfully used proploader.exe -r -D reset=rts -i 10.10.11.156 Welcome.elf
... to load the code.
...with /RTS connected to RESET.

dbetz, 2017-02-12:
I posted what I think is a fix for this problem. It turns out that I never selected the GPIO function of the /RTS pin. That means my reset code probably wasn't really controlling its state. I've fixed that now I think although I still don't pull it high before attempting a reset. I pull it low (which was the state it was in by default) for 10ms and then pull it high. I think that should work. However, we may want to have it high after reset. That means we need to know the reset pin at boot time not just at the moment we want to start loading a program.

dbetz, 2017-02-13:
In addition to the change I mentioned above, I've also made the firmware use the reset-pin setting to initialize the reset pin at power up. I've also added a way to adjust the reset-pin setting on the HTML settings page. Now the reset-pin feature of proploader is probably redundant although it does have a use during serial downloads especially on the RaspberryPi.

PropGit, 2017-02-13:
The loader always sets the reset pin?  Or are you saying it only sets it if the reset-pin option is used (i.e. is assumed to be DTR)?  It seems this needs to be made to either always override the module's setting, or to never override it.

PropGit, 2017-04-10:
Can this issue be closed, or does something still need testing?

We've decided to change RTS functionality to work on CTS instead... but this may still apply in that respect when the change is made.

dbetz, 2017-04-10:
The firmware currently ignores the reset-pin setting sent by proploader and only uses the reset-pin firmware setting to determine which pin to use. Because of this it is able to initialize the pin correctly when the WX boots. I don't think we'll have the problem that Andy reported anymore.

PropGit, 2017-04-10:
Okay, great.  I'll close this issue then.

closed 2017-04-10 by PropGit