parallaxinc/Parallax-ESP #20: Remove Reset-Pin setting from configuration page

PropGit, 2017-02-13:
@dbetz and @AndyLindsay - Unless there's a compelling reason to keep it, please remove the Reset-Pin setting from the configuration page as the loader always takes control and overrides it.

Agreed?

dbetz, 2017-03-13:
I think it would probably be better to have the loader not override this setting and leave it on the configuration page. The reason is that the WX really needs to know this setting at boot time to set the pin to the correct state. It doesn't work as well to set it during the load process.

PropGit, 2017-03-13:
@dbetz - Good point.  Sounds like there's no good reason to keep the override function at all... but rather just have the WX configured for a particular Reset pin and the loader just requests a reset.  Is that what you're suggesting?

PropGit, 2017-05-08:
__Update__ The Reset-Pin setting was retained and updated to include DTR and CTS options.  The CTS option is for use with the SIP form of the module.  A direct CTS to Propeller RESn pin connection is perfect.  The PropLoader v1.0-33 (and later) is a proper fit for the new Wi-Fi firmware Version: v1.0 (2017-04-22 10:25:42 13-gb357dbb).

When using the SIP form, the module _must_ be configured for Reset-Pin: CTS from the configuration page in order for the PropLoader to properly download to the Propeller; ie: PropLoader _does not_ override the Reset-Pin setting anymore.  When using the DIP form, the module _must_ be configured for Reset-Pin: DTR from the configuration page in order for the PropLoader to properly download to the Propeller.

closed 2017-05-08 by PropGit