parallaxinc/Parallax-ESP #28: APSCAN event, count always zero

VonSzarvas, 2018-01-15:
sscp-wifi.c
line: ~16

The event raised to serial always has "count" value set as zero. In the case of an event, "scanCount" is not set.

Example of the event received: !A,0,0
For 2 AP's found, should be: '!A,2,0

DBG output is correct, ESP_TXD output is incorrect (when cmd-events active)

ORIGINAL CODE:

```
static void ICACHE_FLASH_ATTR scan_complete(void *data, int count)
{
    os_printf("scan done callback, found %d\n", count);
    if (flashConfig.sscp_events)
        send_scan_complete_event('!');
    else {
        scanDone = 1;
        scanCount = count;
    }
}
```

POSSIBLE FIX:

```
static void ICACHE_FLASH_ATTR scan_complete(void *data, int count)
{
    os_printf("scan done callback, found %d\n", count);
    
    scanDone = 1;
    scanCount = count;

    if (flashConfig.sscp_events)
        send_scan_complete_event('!');
   
}
```

Note: scanDone flag also moved. Behavior not verified.

PropGit, 2018-01-16:
Thank you for catching this, @VonSzarvas!

@dbetz - Can you verify the suggested fix?

dbetz, 2018-01-17:
I would change it to this:

```
static void ICACHE_FLASH_ATTR scan_complete(void *data, int count)
{
    os_printf("scan done callback, found %d\n", count);
    
    scanCount = count;

    if (flashConfig.sscp_events)
        send_scan_complete_event('!');
    else
        scanDone = 1;
   
}
```

Setting scanDone to 1 will cause the event to be returned when the user calls the POLL function even though it has already been sent as an asynchronous event.

PropGit, 2018-03-22:
Resolved and confirmed by @VonSzarvas

closed 2018-03-22 by PropGit