parallaxinc/Parallax-ESP #29: Build fails with latest pull

VonSzarvas, 2018-03-28:
Something to do with the recent FCount fix?

Here's make output:

```
build/httpd_app.a(sscp.o):(.data+0x7c): undefined reference to `fs_do_finfo'
build/httpd_app.a(sscp.o):(.data+0x84): undefined reference to `fs_do_fcount'
build/httpd_app.a(sscp.o):(.data+0x8c): undefined reference to `fs_do_frun'
collect2: error: ld returned 1 exit status
Makefile.ota:50: recipe for target 'build/httpd.user1.out' failed
make: *** [build/httpd.user1.out] Error 1
```

dbetz, 2018-03-28:
Sorry, I just fixed it.

VonSzarvas, 2018-03-28:
No worries- and very happy you could fix it so quick.

Tested and working great!

dbetz, 2018-03-28:
It wasn’t really a bug to fix. I had forgotten to check in a new file that contained the filesystem API functions.

ClockLoop, 2018-04-01:
Thanks for the work on this dbetz, I was able to build this on a raspberrypi 3, as of today.  (Raspbian Stretch with latest updates)

EXCELLENT!
(i am able to build open spin, and proploader on the pi3 stretch also!   Thanks!(

closed 2018-04-05 by VonSzarvas