parallaxinc/Parallax-ESP #34: Connection issues causes connection leaks

iseries1, 2019-04-29:
Using the tcp connect command cause a connection to be allocated from the pool but if there is a DNS error or connection error this pooled connection gets orphaned.  After 5 bad connection attempts the pool is exhausted and no further connections are possible. 
In sscp-tcp.c there should be sscp_close_connection(c); added before returning an error code to the calling application.

dbetz, 2019-05-16:
I'm working on this now. It's part of a larger update that includes a fix for the flash filesystem overflow issue that was reported.

PropGit, 2019-05-16:
Thanks.  I just noticed the flash filesystem issue was not added to issues (just in email and chat) so I'll make a quick one now that we can reference with the future release.

dbetz, 2019-07-31:
I just pushed a fix for this.

closed 2019-07-31 by dbetz